hello-robot/stretch_ros2 #70: Running colcon test on the Stretch Deep Perception pkg errors

hello-binit, 2023-08-01:
Running `colcon test --packages-select stretch_deep_perception` will yield that it errorred while collecting the package's tests:

```
$ colcon test-result --verbose
build/stretch_deep_perception/pytest.xml: 1 test, 1 error, 0 failures, 0 skipped
- stretch_deep_perception stretch_deep_perception.__init__
  <<< error message
    collection failure
  >>>

```

hello-binit, 2023-08-01:
@hello-atharva, any idea what's going on here?

hello-atharva, 2023-08-02:
I believe that this issue is not `stretch_deep_perception`-specific. Running `colcon test` on `stretch_core` yields the same issue.

hello-atharva, 2023-08-02:
Upon further investigation, I found this
```
hello-robot@stretch-re2-2046:~/test_ws$ cat log/latest_test/stretch_core/stdout.log 
============================= test session starts ==============================
platform linux -- Python 3.10.12, pytest-6.2.5, py-1.11.0, pluggy-0.13.0
cachedir: /home/hello-robot/test_ws/build/stretch_core/.pytest_cache
rootdir: /home/hello-robot/test_ws/src/stretch_ros2/stretch_core
plugins: ament-pep257-0.14.1, ament-lint-0.14.1, ament-flake8-0.14.1, ament-xmllint-0.14.1, launch-testing-2.0.2, launch-testing-ros-0.24.0, ament-copyright-0.14.1, anyio-3.6.2, dash-2.9.3, cov-2.8.1, colcon-core-0.12.1
collecting ... 
collecting 0 items / 1 error                                                   
collected 0 items / 1 error                                                    

==================================== ERRORS ====================================
________________________ ERROR collecting test session _________________________
/usr/lib/python3/dist-packages/pluggy/hooks.py:286: in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
/usr/lib/python3/dist-packages/pluggy/manager.py:92: in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
/usr/lib/python3/dist-packages/pluggy/manager.py:83: in <lambda>
    self._inner_hookexec = lambda hook, methods, kwargs: hook.multicall(
../../../../.local/lib/python3.10/site-packages/_pytest/python.py:200: in pytest_collect_file
    module: Module = ihook.pytest_pycollect_makemodule(path=path, parent=parent)
/usr/lib/python3/dist-packages/pluggy/hooks.py:286: in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
/usr/lib/python3/dist-packages/pluggy/manager.py:92: in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
/usr/lib/python3/dist-packages/pluggy/manager.py:83: in <lambda>
    self._inner_hookexec = lambda hook, methods, kwargs: hook.multicall(
/opt/ros/iron/lib/python3.10/site-packages/launch_testing/pytest/hooks.py:188: in pytest_pycollect_makemodule
    entrypoint = find_launch_test_entrypoint(path)
/opt/ros/iron/lib/python3.10/site-packages/launch_testing/pytest/hooks.py:181: in find_launch_test_entrypoint
    module = path.pyimport()
../../../../.local/lib/python3.10/site-packages/py/_path/local.py:704: in pyimport
    __import__(modname)
../../../../.local/lib/python3.10/site-packages/_pytest/assertion/rewrite.py:170: in exec_module
    exec(co, module.__dict__)
test/test_pep257.py:15: in <module>
    from ament_pep257.main import main
/opt/ros/iron/lib/python3.10/site-packages/ament_pep257/main.py:25: in <module>
    import pydocstyle
../../../../.local/lib/python3.10/site-packages/pydocstyle/__init__.py:1: in <module>
    from .checker import check
../../../../.local/lib/python3.10/site-packages/pydocstyle/checker.py:12: in <module>
    from .config import IllegalConfiguration
../../../../.local/lib/python3.10/site-packages/pydocstyle/config.py:6: in <module>
    from collections import Set, namedtuple
E   ImportError: cannot import name 'Set' from 'collections' (/usr/lib/python3.10/collections/__init__.py)
- generated xml file: /home/hello-robot/test_ws/build/stretch_core/pytest.xml --
=========================== short test summary info ============================
ERROR  - ImportError: cannot import name 'Set' from 'collections' (/usr/lib/p...
!!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!
=============================== 1 error in 0.66s ===============================

```