hello-robot/stretch_ros2 #72: Stretch driver set_mobile_base_velocity_callback can fail to release the robot mode lock

nickswalker, 2023-08-02:
If the base mode is incorrect, this callback returns without releasing the lock, causing other threads which attempt to access the mode lock to wait forever. Lock usage in this file should use the context manager (`self.robot_mode_rwlock.read_access`) provided in the lock implementation to avoid this.

https://github.com/hello-robot/stretch_ros2/blob/333d57dcd60b0e57b520475d452a4579c0a165e8/stretch_core/stretch_core/stretch_driver.py#L77-L87

hello-chintan, 2023-08-19:
Hey @nickswalker, good catch! Thanks for proposing this fix. I'd be happy to merge this in if you would like to raise a PR.

hello-chintan, 2023-09-21:
Just pushed a fix for this. Feel free to reopen if the issue persists.

closed 2023-09-21 by hello-chintan