hello-robot/stretch_ros2 #93: Undefined variables in FUNMAP

benoit-robotics, 2023-11-03:
Two variables are used but are not initialized anywhere in the code, which lead to the funmap node to crash. Tested on ROS 2 Humble.
https://github.com/hello-robot/stretch_ros2/blob/humble/stretch_funmap/stretch_funmap/funmap.py#L1387C14-L1388C32

hello-binit, 2023-11-06:
Hi @benoit-robotics, thanks for reporting the issue. Could you try the fix in the [feature/funmap_rebase branch](https://github.com/hello-robot/stretch_ros2/tree/feature/funmap_rebase)? This should resolve the problem by defining those thread variables.

benoit-robotics, 2023-11-07:
Hi. I tried out this branch but it doesn't work for me. First, the stretch_funmap package contains a COLCON_IGNORE file which needs to be removed. Secondly, the funmap node crashes when I start a headscan : 

> [funmap-8]   File "/home/care-homes/ament_ws/install/stretch_funmap/lib/python3.10/site-packages/stretch_funmap/funmap.py", line 946, in perform_head_scan
> [funmap-8]     head_scan.execute_full(node, fast_scan=fast_scan)
> [funmap-8]   File "/home/care-homes/ament_ws/install/stretch_funmap/lib/python3.10/site-packages/stretch_funmap/mapping.py", line 424, in execute_full
> [funmap-8]     self.execute(head_tilt, far_left_pan, far_right_pan, num_pan_steps, capture_params, node)
> [funmap-8]   File "/home/care-homes/ament_ws/install/stretch_funmap/lib/python3.10/site-packages/stretch_funmap/mapping.py", line 372, in execute
> [funmap-8]     self.capture_point_clouds(node, pose, capture_params)
> [funmap-8]   File "/home/care-homes/ament_ws/install/stretch_funmap/lib/python3.10/site-packages/stretch_funmap/mapping.py", line 344, in capture_point_clouds
> [funmap-8]     cloud_time = node.point_cloud.header.stamp
> [funmap-8] AttributeError: 'NoneType' object has no attribute 'header'