hello-robot/stretch_ros2 #98: Noisy rtabmap recording

xuq1mu, 2024-01-11:
Hi,
I used the `stretch_rtabmap` package to record a map, but the recorded result was quite noisy. When I use the recorded rtabmap to navigate the stretch robot, the robot is often stuck somewhere because of the recorded noise, specifically, the robot is located on an empty plane, but the noisy map makes the robot think obstacles surround it.

If I understand it correctly, the recorded 3D point cloud is down-projected into the 2D cost map for navigation, and the noisy point cloud leads to non-existent obstacles in the 2D cost map.

Is there any filter that could be used during recording? Did you have the same issue with the noisy map?

Thanks in advance!

hello-binit, 2024-01-16:
Hi @xuq1mu, I haven't tried the `stretch_rtabmap` package before. I'd be happy to give it a shot and see if I'm seeing the same noise you're seeing in the map. Would you explain the steps you took and commands you executed to capture the map?

xuq1mu, 2024-01-17:
Thanks for your reply,
I followed the Readme file and used the command `ros2 launch stretch_rtabmap visual_mapping.launch.py teleop_type:=joystick` to capture the map.
Do you think that could be a problem with the resolution of the RealSense camera?

hello-binit, 2024-01-17:
Thanks @xuq1mu. I'll have to try it to know for sure. Could you also send me your robot's serial number (should be printed on a sticker in the trunk of the robot).

xuq1mu, 2024-01-19:
Hi,
after checking the robot's entire body, I still could not find the sticker with the serial number. Could you specify it a little bit more?

xuq1mu, 2024-01-24:
Hi,
after trying, I've found that switching to low resolution and pointing the camera towards the ground can be helpful. By the way, during the navigation with rtabmap, "I found that the robot is still using only 2D lidar to localize itself. Shouldn't it use the camera information to do localization and collision avoidance?

Tsaimingchun14, 2025-10-20:
@xuq1mu Hi, I am also trying to enable camera while navigating. Have you found out how to do so?

hello-robot-shehab, 2025-10-23:
Hi @xuq1mu, sorry that it took us so long to get back to you. The 2D Lidar should not be spinning while running rtabmap, are you still seeing this issue? If you mean during nav2, then correct, the robot uses the 2D lidar for obstacle detection. You can use rtabmap to create a map to use with nav2 though.

Hello @Tsaimingchun14 , sorry what do you mean by "enable camera" while navigating? The camera information should be publishing on the `/camera/color/image_row` ros2 topic. You can visualize this in RViz:

[image]

Sorry you may need to write a quick node to rotate and visualize in an opencv window if you want to. I put together a quick script for you below. You can quickly save it anywhere and run it using `python3 ./show_camera_feed.py`

```
#!/usr/bin/env python3
import rclpy
from rclpy.node import Node
from sensor_msgs.msg import Image
from cv_bridge import CvBridge
import cv2
import numpy as np

class ImageViewer(Node):
    def __init__(self):
        super().__init__('image_viewer')

        self.subscription = self.create_subscription(
            Image,
            '/camera/color/image_raw',
            self.image_callback,
            10)

        self.bridge = CvBridge()
        self.get_logger().info('Image viewer node started, waiting for images...')

    def image_callback(self, msg):
        try:
            # Convert ROS Image message to OpenCV image
            cv_image = self.bridge.imgmsg_to_cv2(msg, desired_encoding='bgr8')
            cv_image =  np.rot90(cv_image, -1)
            cv2.imshow("Camera Feed", cv_image)
            cv2.waitKey(1)
        except Exception as e:
            self.get_logger().error(f'Failed to convert image: {e}')

def main(args=None):
    rclpy.init(args=args)
    node = ImageViewer()
    try:
        rclpy.spin(node)
    except KeyboardInterrupt:
        pass
    finally:
        node.destroy_node()
        cv2.destroyAllWindows()
        rclpy.shutdown()

if __name__ == '__main__':
    main()

``