hello-robot/stretch_ros2 #101: Underestimated URDF mass

marekmlk, 2024-01-17:
Hello!
Summing up all masses in the URDF gives 3.396983 kg while the robot weights around 23 kg. Something clearly is missing and center of gravity as visualized in Gazeo is quite high up causing some issues.

Kind regards,
Marek

hello-binit, 2024-01-17:
Hi @marekmlk, thanks for opening this issue. Are you testing this with Gazebo Classic? We don't have support for a ROS2 Gazebo simulation currently.

marekmlk, 2024-01-18:
Hello @hello-binit, Gazebo Ignition actually. No support for Gazebo is not a problem but do you support URDF? In the end it is general description and simulation agnostic, right? 

I just wondering where is the missing mass. I assume that in the base it must be some kind of counterweight.

hello-binit, 2024-02-01:
Hi @marekmlk, good point. Take a look at this post: [URDF Mass Parameters](https://forum.hello-robot.com/t/urdf-mass-parameters/789). It should help you find a up-to-date URDF that has accurate mass and inertia data.

marekmlk, 2024-02-01:
Yeah, this data looks correct. Thank you! Would be great to port it to ros2 though :)

hello-binit, 2024-02-01:
Ah good call! I'll leave this issue open as a reminder to port the accurate URDF data PR to ROS2.