hello-robot/stretch_ros2 #114: Missing numpy dependency fo stretch_funmap

marekmlk, 2024-03-13:
Hello!

[stretch_funmap](https://github.com/hello-robot/stretch_ros2/blob/humble/stretch_funmap/stretch_funmap/cython_min_cost_path.pyx#L1-L6) uses `numpy` and `cython` but they are included in [package.xml](https://github.com/hello-robot/stretch_ros2/blob/humble/stretch_funmap/package.xml).

Without that information `rosdep` fails to install all dependencies properly.

Kind regards,
Marek

hello-binit, 2024-03-20:
Hi @marekmlk, thanks for reporting this. I'd expect Numpy and Cython to be installed on a robot that has already gone through a Stretch Installation (includes [cython](https://github.com/hello-robot/stretch_install/blob/bde279cd8925379148e579fa16ea3fd9d3c5c88c/factory/22.04/stretch_install_system.sh#L55)), which includes installing the Stretch Body Python library (includes [numpy](https://github.com/hello-robot/stretch_body/blob/master/body/setup.py#L22C24-L22C37)).

Have you had issues on a Stretch where these dependencies were missing?

marekmlk, 2024-03-20:
Thanks for the reply!

No, I did not faced this issue on the robot but in simulation. I'm relying on rosdep to install all dependencies in Docker container and I stumped upon it. 

You are right, `cyton` and `numpy`are so popular so it is very unlikely that they are not on the system. However, when building Docker image from scratch it may happen. 

I also think that it is good practice to include all possible dependencies in `package.xml` :)

Kind regards,
Marek