hello-robot/stretch_ros2 #159: Double try-catch

hello-binit, 2024-08-27:
Within Stretch Driver's loop, there's a double try-catch. If a Ctrl-C appears, which "catch" logic get triggered? Does everything get shutdown properly?

[image]

Shakthi-Bala, 2026-01-19:
Hi,
To my understanding When Ctrl-C is pressed while *executer.spin()* is running, Python executes inner finally first before going to *except* block, So the node is destroyed before stopping the robot, This could make the *stop* command not being sent reliably, Which means robot would do whatever the last command was unless explicitly *stop* command is sent, Instead the *stop* command can be written in inner *finally* itself.

hello-binit, 2026-01-21:
Hi @Shakthi-Bala , thank you for your comment! I didn't expect anyone to look at this. I agree with your assessment about `stop()` potentially being missed. I plan to add an unit test and move `stop()` to the inner logic if needed. Let me know if you'd like to take a crack at it instead.

hello-binit, 2026-01-28:
That sounds like a good plan. Perhaps `robot` and `gamepad_teleop` is mocked in this unit test to verify that the stop() methods are actually called.