hello-robot/stretch_ros2 #192: grasp_object error

hyuhang13, 2025-04-01:
Hello! When I tried to run ros2 launch stretch_demos grasp_object.launch.py, an error occurred after the object was detected. I have tried the script, but it still doesn’t work. May I ask what happened?
I hope you can help me answer, thank you!
```
hello-robot@stretch-se3-3102:~/ament_ws$ ros2 launch stretch_demos grasp_object.launch.py 
[INFO] [launch]: All log files can be found below /home/hello-robot/.ros/log/2025-04-01-05-44-45-050106-stretch-se3-3102-7858
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: D435i launched in high resolution
[INFO] [realsense2_camera_node-1]: process started with pid [7860]
[INFO] [stretch_driver-5]: process started with pid [7868]
[INFO] [d435i_accel_correction-2]: process started with pid [7862]
[INFO] [joint_state_publisher-3]: process started with pid [7864]
[INFO] [robot_state_publisher-4]: process started with pid [7866]
[INFO] [funmap-6]: process started with pid [7870]
[INFO] [grasp_object-7]: process started with pid [7872]
[INFO] [static_transform_publisher-8]: process started with pid [7874]
[static_transform_publisher-8] [INFO] [1743511485.229607637] [static_transform_publisher_mvr5SRaR1TUpXZS7]: Spinning until stopped - publishing transform
[static_transform_publisher-8] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-8] rotation: ('0.500000', '0.500000', '-0.500000', '0.500000')
[static_transform_publisher-8] from 'link_straight_gripper' to 'link_straight_gripper_aligned'
[d435i_accel_correction-2] [INFO] [1743511485.751144308] [d435i_accel_correction_node]: d435i_accel_correction_node started
[realsense2_camera_node-1] [INFO] [1743511485.933741490] [camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1743511485.933844475] [camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1743511485.933855178] [camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1743511486.041612379] [camera]: Device with serial number 250122075290 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1743511486.041702444] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:0d.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1743511486.041721637] [camera]: Device with name Intel RealSense D435IF was found.
[realsense2_camera_node-1] [INFO] [1743511486.041901836] [camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1743511486.041925306] [camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1743511486.041938398] [camera]: Resetting device...
[grasp_object-7] [INFO] [1743511486.399886003] [stretch_funmap]: MoveBase __init__: self.debug_directory = None
[grasp_object-7] [INFO] [1743511486.442579906] [grasp_object]: grasp_object started
[stretch_driver-5] [INFO] [1743511486.574775608] [stretch_driver]: For use with S T R E T C H (TM) RESEARCH EDITION from Hello Robot Inc.
[stretch_driver-5] [INFO] [1743511486.575049888] [stretch_driver]: stretch_driver started
[funmap-6] [INFO] [1743511487.226339477] [funmap]: funmap started
[stretch_driver-5] Key mapped to End-Of-Arm Tool: eoa_wrist_dw3_tool_sg3
[stretch_driver-5] [INFO] [1743511488.297759503] [stretch_driver]: Changed to mode = position
[stretch_driver-5] [INFO] [1743511488.299451593] [stretch_driver]: broadcast_odom_tf = True
[stretch_driver-5] [INFO] [1743511488.327101227] [stretch_driver]: rate = 30.0 Hz
[stretch_driver-5] [INFO] [1743511488.330223024] [stretch_driver]: twist timeout = 0.5 s
[stretch_driver-5] [INFO] [1743511488.332149994] [stretch_driver]: base_frame_id = base_link
[stretch_driver-5] [INFO] [1743511488.333829564] [stretch_driver]: odom_frame_id = odom
[grasp_object-7] [INFO] [1743511488.483369271] [grasp_object]: Node grasp_object connected to /home_the_robot service.
[grasp_object-7] [INFO] [1743511488.484494359] [grasp_object]: Node grasp_object connected to /stow_the_robot service.
[grasp_object-7] [INFO] [1743511488.485868545] [grasp_object]: Node grasp_object connected to /stop_the_robot service.
[grasp_object-7] [INFO] [1743511488.487097863] [grasp_object]: Node grasp_object connected to /switch_to_trajectory_mode service.
[grasp_object-7] [INFO] [1743511488.492077098] [grasp_object]: Node grasp_object connected to /switch_to_position_mode service.
[grasp_object-7] [INFO] [1743511488.494246753] [grasp_object]: Node grasp_object connected to /switch_to_navigation_mode service.
[grasp_object-7] [INFO] [1743511488.494623514] [grasp_object]: Using the following directory for debugging files: /home/hello-robot/stretch_user/debug/
[grasp_object-7] [INFO] [1743511488.496999069] [grasp_object]: Waiting for /funmap/trigger_reach_until_contact' service
[funmap-6] [INFO] [1743511488.523573086] [funmap]: Node funmap connected to /home_the_robot service.
[funmap-6] [INFO] [1743511488.524851038] [funmap]: Node funmap connected to /stow_the_robot service.
[funmap-6] [INFO] [1743511488.528046096] [funmap]: Node funmap connected to /stop_the_robot service.
[funmap-6] [INFO] [1743511488.529353169] [funmap]: Node funmap connected to /switch_to_trajectory_mode service.
[funmap-6] [INFO] [1743511488.530427523] [funmap]: Node funmap connected to /switch_to_position_mode service.
[funmap-6] [INFO] [1743511488.531801789] [funmap]: Node funmap connected to /switch_to_navigation_mode service.
[funmap-6] [INFO] [1743511488.532064656] [funmap]: Node funmap waiting to receive first point cloud.
[realsense2_camera_node-1] [INFO] [1743511492.130483802] [camera]: Device with serial number 250122075290 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1743511492.130549741] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:0d.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1743511492.130563547] [camera]: Device with name Intel RealSense D435IF was found.
[realsense2_camera_node-1] [INFO] [1743511492.130775869] [camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1743511492.130793232] [camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1743511492.130845786] [camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1743511492.488820962] [camera]: JSON file is loaded! (/home/hello-robot/ament_ws/install/stretch_core/share/stretch_core/config/HighAccuracyPreset.json)
[realsense2_camera_node-1] [INFO] [1743511492.488893856] [camera]: Device Name: Intel RealSense D435IF
[realsense2_camera_node-1] [INFO] [1743511492.488903843] [camera]: Device Serial No: 250122075290
[realsense2_camera_node-1] [INFO] [1743511492.488912199] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:0d.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1743511492.488922692] [camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1743511492.488931295] [camera]: Device Product ID: 0x0B3A
[... 82 lines cut ...]
[grasp_object-7]      voi.x_in_m = 0.15
[grasp_object-7]      voi.y_in_m = 0.4
[grasp_object-7]      voi.z_in_m = 0.07
[grasp_object-7] MaxHeightImage information:
[grasp_object-7]      image.shape = (434, 350)
[grasp_object-7]      image.dtype = uint8
[grasp_object-7]      m_per_pix = 0.006
[grasp_object-7]      m_per_height_unit = 0.004566929133858267
[grasp_object-7]      voi.x_in_m = 2.1
[grasp_object-7]      voi.y_in_m = 2.6
[grasp_object-7]      voi.z_in_m = 1.16
[grasp_object-7] MaxHeightImage information:
[grasp_object-7]      image.shape = (334, 434)
[grasp_object-7]      image.dtype = uint8
[grasp_object-7]      m_per_pix = 0.006
[grasp_object-7]      m_per_height_unit = 0.004566929133858267
[grasp_object-7]      voi.x_in_m = 2.6
[grasp_object-7]      voi.y_in_m = 2.0
[grasp_object-7]      voi.z_in_m = 1.16
[grasp_object-7] MaxHeightImage saving to base_filename = /home/hello-robot/stretch_user/debug/grasp_object/look_at_surface_20250401054510
[grasp_object-7] Finished saving.
[grasp_object-7] segment_max_height_image.py : find_floor
[stretch_driver-5] New collision pair event: link_gripper_s3_body_TO_link_head_tilt [1743511519.661856]
[INFO] [grasp_object-7]: process has finished cleanly [pid 7872]
```

hello-fazil, 2025-04-07:
Hey this issue is related to this issue in Stretch funmap module. The fix should be pushed soon. 
https://github.com/hello-robot/stretch_ros2/issues/193

hyuhang13, 2025-04-08:
> Hey this issue is related to this issue [#193](https://github.com/hello-robot/stretch_ros2/issues/193) in Stretch funmap module. We are working on the fix and will be pushed soon. Thankyou for reporting.嘿，这个问题与Stretch funmap模块中的 [#193](https://github.com/hello-robot/stretch_ros2/issues/193) 相关。我们正在修复，很快就会发布。谢谢你的报道。

Ok, thank you for your concern. I will wait patiently.

closed 2025-04-21 by hello-fazil